BLE-Research-Group/MetaRadar #15: Crashes on start

dgudim, 2023-03-01:
```FATAL EXCEPTION: main
Process: f.cking.software, PID: 9624
java.lang.RuntimeException: java.lang.reflect.InvocationTargetException
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:581)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067)
Caused by: java.lang.reflect.InvocationTargetException
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:571)
	... 1 more
Caused by: l8.c: Could not create instance for [Factory:'r6.u']
	at m8.c.a(Unknown Source:185)
	at m8.a.c(Unknown Source:0)
	at s8.b.b(Unknown Source:92)
	at s8.b.a(Unknown Source:153)
	at g8.c.b(Unknown Source:30)
	at androidx.lifecycle.f0.b(Unknown Source:61)
	at androidx.lifecycle.f0.a(Unknown Source:12)
	at b0.l4.c0(Unknown Source:31)
	at r6.f.b(Unknown Source:75)
	at m6.c$a.X(Unknown Source:37)
	at l0.a.a(Unknown Source:49)
	at l0.a.X(Unknown Source:8)
	at d6.e$a.b0(Unknown Source:39)
	at l0.a.b0(Unknown Source:57)
	at f.cking.software.ui.a.b0(Unknown Source:83)
	at l0.a.b0(Unknown Source:57)
	at b0.h2.a(Unknown Source:50)
	at b0.f2.b0(Unknown Source:32)
	at l0.a.b0(Unknown Source:57)
	at e0.l0.a(Unknown Source:28)
	at b0.o5.a(Unknown Source:92)
	at b0.g2$a.b0(Unknown Source:46)
	at l0.a.b0(Unknown Source:57)
	at e0.l0.a(Unknown Source:28)
	at b0.g2.a(Unknown Source:816)
	at f.cking.software.ui.MainActivity$b.b0(Unknown Source:99)
	at l0.a.b0(Unknown Source:57)
	at androidx.compose.ui.platform.o0.a(Unknown Source:23)
	at androidx.compose.ui.platform.a$a.b0(Unknown Source:30)
	at l0.a.b0(Unknown Source:57)
	at e0.l0.a(Unknown Source:28)
	at androidx.compose.ui.platform.p0.a(Unknown Source:329)
	at androidx.compose.ui.platform.x$i.b0(Unknown Source:40)
	at l0.a.b0(Unknown Source:57)
	at e0.l0.a(Unknown Source:28)
	at androidx.compose.ui.platform.x.a(Unknown Source:542)
	at androidx.compose.ui.platform.n2.b0(Unknown Source:34)
	at l0.a.b0(Unknown Source:57)
	at e0.l0.a(Unknown Source:28)
	at androidx.compose.ui.platform.o2.b0(Unknown Source:159)
	at l0.a.b0(Unknown Source:57)
	at e0.o.y(Unknown Source:22)
	at x8.a.q0(Unknown Source:42)
	at e0.i.Q(Unknown Source:119)
	at e0.i.M(Unknown Source:13)
	at e0.i0.y(Unknown Source:17)
	at e0.y1.a(Unknown Source:44)
	at e0.i0.s(Unknown Source:12)
	at androidx.compose.ui.platform.WrappedComposition$a.e0(Unknown Source:65)
	at androidx.compose.ui.platform.AndroidComposeView.setOnViewTreeOwnersAvailable(Unknown Source:11)
	at androidx.compose.ui.platform.WrappedComposition.s(Unknown Source:12)
	at androidx.compose.ui.platform.WrappedComposition.l(Unknown Source:18)
	at androidx.lifecycle.o$a.a(Unknown Source:17)
	at androidx.lifecycle.o.a(Unknown Source:114)
	at androidx.compose.ui.platform.WrappedComposition$a.e0(Unknown Source:34)
	at androidx.compose.ui.platform.AndroidComposeView.onAttachedToWindow(Unknown Source:103)
	at android.view.View.dispatchAttachedToWindow(View.java:22468)
	at android.view.ViewGroup.dispatchAttachedToWindow(ViewGroup.java:3686)
	at android.view.ViewGroup.dispatchAttachedToWindow(ViewGroup.java:3695)
	at android.view.ViewGroup.dispatchAttachedToWindow(ViewGroup.java:3695)
	at android.view.ViewGroup.dispatchAttachedToWindow(ViewGroup.java:3695)
	at android.view.ViewGroup.dispatchAttachedToWindow(ViewGroup.java:3695)
	at android.view.ViewGroup.dispatchAttachedToWindow(ViewGroup.java:3695)
	at android.view.ViewGroup.dispatchAttachedToWindow(ViewGroup.java:3695)
	at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:3544)
	at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:2991)
	at android.view.ViewRootImpl$TraversalRunnable.run(ViewRootImpl.java:10665)
	at android.view.Choreographer$CallbackRecord.run(Choreographer.java:1301)
	at android.view.Choreographer$CallbackRecord.run(Choreographer.java:1309)
	at android.view.Choreographer.doCallbacks(Choreographer.java:923)
	at android.view.Choreographer.doFrame(Choreographer.java:852)```

Semper-Viventem, 2023-03-01:
Let's see. This log is from the obfuscated release build. Unfortunately, I don't have a mapping file for this app version, it would be helpful if you could share the crash stacktrace from the [debug](https://github.com/Semper-Viventem/MetaRadar/releases) build.

dgudim, 2023-03-01:
Yup. Here. 
However, i found that it only crashes if the bluetooth is off when the app is opened.

```FATAL EXCEPTION: main
Process: f.cking.software.debug, PID: 19052
java.lang.RuntimeException: java.lang.reflect.InvocationTargetException
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:581)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067)
Caused by: java.lang.reflect.InvocationTargetException
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:571)
	... 1 more
Caused by: org.koin.core.error.InstanceCreationException: Could not create instance for [Factory:'f.cking.software.ui.main.MainViewModel']
	at org.koin.core.instance.InstanceFactory.create(InstanceFactory.kt:61)
	at org.koin.core.instance.FactoryInstanceFactory.get(FactoryInstanceFactory.kt:38)
	at org.koin.core.registry.InstanceRegistry.resolveInstance$koin_core(InstanceRegistry.kt:111)
	at org.koin.core.scope.Scope.resolveValue(Scope.kt:237)
	at org.koin.core.scope.Scope.resolveInstance(Scope.kt:224)
	at org.koin.core.scope.Scope.get(Scope.kt:205)
	at org.koin.androidx.viewmodel.factory.KoinViewModelFactory.create(KoinViewModelFactory.kt:36)
	at androidx.lifecycle.ViewModelProvider.get(ViewModelProvider.kt:187)
	at androidx.lifecycle.ViewModelProvider.get(ViewModelProvider.kt:153)
	at org.koin.androidx.viewmodel.GetViewModelKt.resolveViewModel(GetViewModel.kt:43)
	at f.cking.software.ui.main.MainScreen.Screen(MainScreen.kt:267)
	at f.cking.software.ui.ComposableSingletons$ScreenNavigationCommandsKt$lambda-1$1.invoke(ScreenNavigationCommands.kt:18)
	at f.cking.software.ui.ComposableSingletons$ScreenNavigationCommandsKt$lambda-1$1.invoke(ScreenNavigationCommands.kt:18)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:116)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:34)
	at f.cking.software.common.navigation.Navigator$handle$1.invoke(Navigator.kt:22)
	at f.cking.software.common.navigation.Navigator$handle$1.invoke(Navigator.kt:22)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:107)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:34)
	at f.cking.software.ui.MainActivity$onCreate$1$1.invoke(MainActivity.kt:65)
	at f.cking.software.ui.MainActivity$onCreate$1$1.invoke(MainActivity.kt:58)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:107)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:34)
	at androidx.compose.material.MaterialTheme_androidKt.PlatformMaterialTheme(MaterialTheme.android.kt:23)
	at androidx.compose.material.MaterialThemeKt$MaterialTheme$1$1.invoke(MaterialTheme.kt:82)
	at androidx.compose.material.MaterialThemeKt$MaterialTheme$1$1.invoke(MaterialTheme.kt:81)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:107)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:34)
	at androidx.compose.runtime.CompositionLocalKt.CompositionLocalProvider(CompositionLocal.kt:228)
	at androidx.compose.material.TextKt.ProvideTextStyle(Text.kt:265)
	at androidx.compose.material.MaterialThemeKt$MaterialTheme$1.invoke(MaterialTheme.kt:81)
	at androidx.compose.material.MaterialThemeKt$MaterialTheme$1.invoke(MaterialTheme.kt:80)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:107)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:34)
	at androidx.compose.runtime.CompositionLocalKt.CompositionLocalProvider(CompositionLocal.kt:228)
	at androidx.compose.material.MaterialThemeKt.MaterialTheme(MaterialTheme.kt:72)
	at f.cking.software.ui.MainActivity$onCreate$1.invoke(MainActivity.kt:49)
	at f.cking.software.ui.MainActivity$onCreate$1.invoke(MainActivity.kt:47)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:107)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:34)
	at androidx.compose.ui.platform.ComposeView.Content(ComposeView.android.kt:404)
	at androidx.compose.ui.platform.AbstractComposeView$ensureCompositionCreated$1.invoke(ComposeView.android.kt:250)
	at androidx.compose.ui.platform.AbstractComposeView$ensureCompositionCreated$1.invoke(ComposeView.android.kt:249)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:107)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:34)
	at androidx.compose.runtime.CompositionLocalKt.CompositionLocalProvider(CompositionLocal.kt:228)
	at androidx.compose.ui.platform.CompositionLocalsKt.ProvideCommonCompositionLocals(CompositionLocals.kt:177)
	at androidx.compose.ui.platform.AndroidCompositionLocals_androidKt$ProvideAndroidCompositionLocals$3.invoke(AndroidCompositionLocals.android.kt:123)
	at androidx.compose.ui.platform.AndroidCompositionLocals_androidKt$ProvideAndroidCompositionLocals$3.invoke(AndroidCompositionLocals.android.kt:122)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:107)
	at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jvm.kt:34)
	at androidx.compose.runtime.CompositionLocalKt.CompositionLocalProvider(CompositionLocal.kt:228)
[... cut ...]
	at org.koin.core.instance.InstanceFactory.create(InstanceFactory.kt:54)
	... 105 more
Caused by: java.lang.NullPointerException: bluetoothAdapter.bluetoothLeScanner must not be null
	at f.cking.software.data.helpers.BleScannerHelper.<init>(BleScannerHelper.kt:38)
	at f.cking.software.data.DataModule$module$1$1.invoke(DataModule.kt:15)
	at f.cking.software.data.DataModule$module$1$1.invoke(DataModule.kt:15)
	at org.koin.core.instance.InstanceFactory.create(InstanceFactory.kt:54)
	... 117 more```

Semper-Viventem, 2023-03-01:
Thank you for the crash log. I've just merged the potential fix for this issue. Could you please check the [0.6-alpha](https://github.com/Semper-Viventem/MetaRadar/releases/tag/20230301213716) build and check if the crash is gone?

dgudim, 2023-03-02:
Nice, it's now fixed

closed 2023-03-02 by dgudim