BLE-Research-Group/MetaRadar #17: Some devices not showing

yazae, 2023-03-01:
Hi, I don’t really understand why but I have 3 paired bluetooth devices who don’t show up in the extensive list your app is showing, actually no devices I use appear and only some Apple and Microsoft ones without names… it’s a bit confusing, do you know why it could be the case ?

I was finding your app useful in case of lost headphone or this kind of things but if it doesn’t show them in the first case, it’s a bit weird ^^

Semper-Viventem, 2023-03-01:
Hey, could you clarify if you run the app on the same smartphone that your devices are paired with? There are scenarios for privacy protection for BLE devices, one of them is a randomized resolvable address when only the master device that pared with your wearable device may resolve the real name and address of the BLE device. It would be helpful if you can also share more information about the devices you trying to find.

yazae, 2023-03-01:
Oh yes, sorry I was not going to post an issue in the first place but there wasn’t a "chat" section on this project

So : 
- Yupp the devices I talk about are paired on the same phone that the app
- I’m on Android 11, /e/OS
- I was trying to find some samsung airbuds, my other android phone, and a bluetooth headphone (seineiser I think ?)

Semper-Viventem, 2023-03-01:
Hmm, I successfully found my Samsung headphones, they even don't have an address-protection. I'll check what's going on, but unfortunately, I don't have any /e/OS devices and the issue may be related to the system's specific behavior

yazae, 2023-03-01:
that’s cool, it may be that I didn’t authorize your app the good way or another thing
but since I still detect some devices, it’s a bit weird :/

I can do some checks on my phone if you think of anything.

Semper-Viventem, 2023-03-01:
We can check if other BLE apps may see your devices nearby, could you check the visibility issue with other scanner apps? ([This one](https://play.google.com/store/apps/details?id=com.macdom.ble.blescanner) is very good)

yazae, 2023-03-11:
hmm it seems to be the same ?
I have a question who may be stupid, but do your app handle other bluetooth devices than BLE ones ?
I don’t fully understand what is BLE against "normal" bluetooth, but I have no idea how to know what my devices are using ^^’

Semper-Viventem, 2023-03-17:
Hey, BLE and Bluetooth are two different protocols, that are not related to each other except for the fact that both are maintained by Bluetooth SIG. At the moment the app doesn't support a canonical Bluetooth and analyzes only BLE to avoid hight consuming battery problems

Semper-Viventem, 2023-03-27:
I'll close the issue, it looks like the issue is not related to the wrong app behavior. But I'll consider if it is possible to support Bluetooth in the radar in future

closed 2023-03-27 by Semper-Viventem