BLE-Research-Group/MetaRadar #28: The app is crashing

userkilled, 2023-03-02:
It absolutely crashes when you press the 'scan' button if one of the interfaces needed by the application is turned off. And always asks for access to the location if 'permission during use'

Semper-Viventem, 2023-03-02:
> App crashes if one of the interfaces is turned off

Could you specify the app version? There were a few fixes regarding this behavior today.

<br>

> Always asks for location even if `permission during use` is granted

The app scans BLE in the background, so the background location access is necessary for correct service working

userkilled, 2023-03-02:
0.4 alfa

Okay, I'll keep that in mind.
Thanks. Probably should tell the location to be in any mode. 

Oh, that's right, after reading normally I saw that from all sources.

userkilled, 2023-03-02:
Я продублировал, выходит. Извините

closed 2023-03-02 by userkilled

userkilled, 2023-03-02:
Только что обновил, всё прекрасно ☕

Semper-Viventem, 2023-03-02:
Класс, все равно спасибо за репорт!