BLE-Research-Group/MetaRadar #43: Can you please add a tag and rename the device?

I want to tag some devices as home, office , car and I also want to rename some devices to more rememberable like Office TV

VarunBatraIT, 2023-04-21:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Semper-Viventem, 2023-04-21:
Hey, that's a really good idea, I just decided to implement this a few days ago because I also faced such a problem. Technically, it's implemented at the data layer and waiting for UI implementation!

Semper-Viventem, 2023-04-23:
Hey,
Device tags will be available since app version `0.17-beta`. I'm not sure about custom names but will add a comment field in the future.

https://github.com/Semper-Viventem/MetaRadar/pull/45

closed 2023-04-23 by Semper-Viventem