BLE-Research-Group/MetaRadar #51: Hungarian translation: strings.xml in attachment

kekmacska, 2023-06-17:
Dear Konstantin Kulikov / Semper-Viventem!

I'm contacting you, because I have translated your app, BLE Radar to my language, hungarian. I will send the necessary [strings.xml](https://www.mediafire.com/file/6ys94w3tl0bzbec/strings.xml/file) file as a mediafire link (Upload region is France, because I have an always-on VPN), since I can't upload it to Github directly. Please, if you find it worthy, implement it into the app officially. Also, if you find any errors that I have made, please let me know.

                                                 I'm looking forward to your reply,
                                                                                      kekmacska

Semper-Viventem, 2023-06-19:
Hey, so cool, I'll support your Hungarian translation as soon as possible.

Semper-Viventem, 2023-07-04:
I've added your localized strings as a part of this Pull-Request: https://github.com/Semper-Viventem/MetaRadar/issues/51

Semper-Viventem, 2023-07-04:
The localization delivery version is `0.18.0-beta`

closed 2023-07-04 by Semper-Viventem