BLE-Research-Group/MetaRadar #57: Devices do not show

natrius, 2023-08-29:
**Describe the bug**
Well, no devices are shown but in the draw-down menu of android it shows that there are 7 devices found. I can click on them even if i don't see them. 

**Expected behavior**
A list of devices.

**Screenshots**

[image]

**Smartphone (please complete the following information):**
 - Device: Xiaomi Mi 9
 - OS: Android 10
 - Browser: App
 - Version: 0.18.1-beta

I basically just need to scan my own earbuds to get them implemented somewhere else.

Semper-Viventem, 2023-08-29:
Hey, could you please
* If there are any error messages in the Journal tab
* If location is enabled for your device and granted for the app (location is needed for Bluetooth according to the Android Privacy Policies)

Manamama, 2023-09-29:
The App is amazing as I needed just this: I thought of coding it myself, but then found it in Github here, ready. Kudos. 

A tip re the above. Not sure if a bug or a feature, but with Location Always On disabled: 
[image] it does not scan: 
[image]
only if fully and always enabled, it does: 
[image]
(NRF Connect, EFR Connect etc do not need it always on.)

Semper-Viventem, 2024-02-23:
Hey, the app had quite a lot of updates since the report. I'll close the issue as the problem is not reproducible for me.

Responding to the question from @Manamama, the background location is required for radar working in the background because of the android security policies. We cannot obtain Bluetooth without the location permission.

closed 2024-02-23 by Semper-Viventem

Manamama, 2024-02-26:
Quick FYI that before and after the most recent update, 0.23.0 beta, it has been working fine - two standard stock,  rooted, Oppo Droids.

natrius, 2024-02-26:
I would like to add a possible culprit for such visibility-issues: Xiaomi has some kind of automatic dark-mode even for apps that don't support it. That made some problems with another app already. 

BLE Radar works good right now, everything visible.