BLE-Research-Group/MetaRadar #70: Package installer is crashing under Android 14 on Pixel 6 pro on every try installing update.

p-f-w, 2023-12-21:
Maybe not really caused by "Metaradar", other Apps like "SyncThing", and there is a link in issues to "cromite", also affected by crashing installer.

Semper-Viventem, 2023-12-22:
I wasn't able to reproduce the problems during my tests. Could you try to remove and install the app (not to update)? If you have some data collected don't forget to backup the database in the settings.

0b-1, 2024-01-02:
I have same problem on Pixel7a. Install from IzzyDroid and apk both crash. I can't remove app as it is first installation attempt. Not sure if below log is related:

`FATAL EXCEPTION: main
Process: com.google.android.packageinstaller, PID: 32545
java.lang.RuntimeException: Could not copy bitmap to parcel blob.
	at android.graphics.Bitmap.nativeWriteToParcel(Native Method)
	at android.graphics.Bitmap.writeToParcel(Bitmap.java:2271)
	at android.os.Parcel.writeParcelable(Parcel.java:2584)
	at com.android.packageinstaller.PackageUtil$AppSnippet.writeToParcel(PackageUtil.java:151)
	at android.os.Parcel.writeParcelable(Parcel.java:2584)
	at android.os.Parcel.writeValue(Parcel.java:2485)
	at android.os.Parcel.writeValue(Parcel.java:2362)
	at android.os.Parcel.writeArrayMapInternal(Parcel.java:1298)
	at android.os.BaseBundle.writeToParcelInner(BaseBundle.java:1843)
	at android.os.Bundle.writeToParcel(Bundle.java:1389)
	at android.os.Parcel.writeBundle(Parcel.java:1367)
	at android.content.Intent.writeToParcel(Intent.java:11807)
	at android.os.Parcel.writeTypedObject(Parcel.java:2203)
	at android.app.IActivityTaskManager$Stub$Proxy.startActivity(IActivityTaskManager.java:2077)
	at android.app.Instrumentation.execStartActivity(Instrumentation.java:1873)
	at android.app.Activity.startActivityForResult(Activity.java:5615)
	at android.app.Activity.startActivityForResult(Activity.java:5573)
	at android.app.Activity.startActivity(Activity.java:6071)
	at android.app.Activity.startActivity(Activity.java:6038)
	at com.android.packageinstaller.PackageInstallerActivity.startInstall(PackageInstallerActivity.java:701)
	at com.android.packageinstaller.PackageInstallerActivity.lambda$bindUi$0(PackageInstallerActivity.java:479)
	at com.android.packageinstaller.PackageInstallerActivity.$r8$lambda$RCIm8wl1VPqdfQgkgmrBIDbuvOQ(PackageInstallerActivity.java:0)
	at com.android.packageinstaller.PackageInstallerActivity$$ExternalSyntheticLambda0.onClick(R8$$SyntheticClass:0)
	at com.android.packageinstaller.AlertController$ButtonHandler.handleMessage(AlertController.java:144)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loopOnce(Looper.java:205)
	at android.os.Looper.loop(Looper.java:294)
	at android.app.ActivityThread.main(ActivityThread.java:8194)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:552)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:971)`

Semper-Viventem, 2024-01-03:
It looks like the problem is not in the app but in the app installer. But I have an idea why it might crash the app installer. Would be nice if you could check installation for the release version `0.21.4-beta`, available on the [releases page](https://github.com/Semper-Viventem/MetaRadar/releases?q=release+build&expanded=true)

0b-1, 2024-01-03:
Thank you, tested with v0.21.4-beta and it crashes as well.

Semper-Viventem, 2024-01-03:
Ok, one more issue candidate has been fixed. Try `0.21.5-beta` from releases. Hope it will help

0b-1, 2024-01-05:
I tested again with 0.21.5-beta and it is still not installing as well  (no message, it seems it just doesn't do anything).

0b-1, 2024-01-05:
When trying to install the apk I was using Material Files as manager. I just re-activated 'Files by Google', opened it and launched the apk. I had to "allow from this source", then relaunched and it is now installed.  Weird.
And from F-droid it wasn't installing as well.

Semper-Viventem, 2024-01-06:
As I see this is not a problem with the BLE Radar application packaging but something is wrong with Android 14 package installer. The problem happens not only with the Radar app but with other APKs even in the F-Droid store. I'll see what can we do here but I guess it should be fixed on the AOSP side.

p-f-w, 2024-01-20:
the creator of "TinyWeatherForecastGermany"

https://codeberg.org/Starfish/TinyWeatherForecastGermany

has changed something and version 0.61.4 could be now installed here

"added workaround to prevent the package installer from crashing when installing and/or updating the app"

Semper-Viventem, 2024-01-20:
@p-f-w looks like a good reference. I'll try to do the same fix as here: https://codeberg.org/Starfish/Imagepipe/commit/19ddd2c2390c59b75b41f274990ace6cbf2a143a

p-f-w, 2024-02-22:
i close the issue, since i have installed 0.22.8-beta with "App manager" from F-Droid successfully further installations of 0.22.9-beta and 0.23.0-beta where here ok with the normal paket installer from pixel

closed 2024-02-22 by p-f-w