BLE-Research-Group/MetaRadar #90: minor issues with fastlane graphics

IzzySoft, 2024-01-18:
Just to point out two minor issue with your fastlane graphics:

* screenshots have the wrong proportions (they are about twice as high as they should be)
* if your `header.png` was named `featureGraphic.png` it could be used as such, proportions match the specs exactly :wink: 

Thanks for checking (and hopefully adjusting)!

Semper-Viventem, 2024-01-20:
Hey, thank you for noticing @IzzySoft. Could you specify the correct size of the screenshots? I'll re-do all of them soon anyway

IzzySoft, 2024-01-21:
Oof. I don't know the correct *size* – just that there's a check about *proportions* which yours currently don't meet. Let me ask the all-knowing Internet for you … hm, seems if it's all-knowing it's not all-finding. It was something like 1:2, or between 1:1.7 and 1:2 as far as I remember. PlayStore seems to want 16:9 or 9:16 according to one documentation I found, which falls into that range (1:1.77) – another mentions "The maximum screenshot dimension can't be more than twice as long as the minimum dimension (2:1 aspect ratio)" – which again matches. So that should be the safe range.

Another hint I found in several notes: "Min length for any side: 320px, Max length for any side: 3840px, Max aspect ratio: 2:1."

Semper-Viventem, 2024-01-21:
Hey, I've updated screenshots and `featureGraphic`, so they should be fine now. Thank you for your help!

IzzySoft, 2024-01-21:
featureGraphic looks good! But what happened with the aspect ratio of the screenshots? (picking up the calc) `2400/1080` hm, that thing says `2,222222222`. Need a mathematician here. Is that between 1.7 and 2.0? Guess one could convince bard and GPT to confirm – but could you please check again? :see_no_evil:

Apologies for the phrasing, but I could not resist… They unfortunately do not meet the specs still, though it's better than before. Do you think you could fix that – or will you let it stay at that? While "1:2 pics" might not fill a screen that has an 1:3 ratio in height, at least all details would be visible on screens with a ratio of 1:2 then. The 2.22222 ratio pics would need to decreased in size to fit the height, and thus get hard to read. Not sure if apps would resize or scroll then, though, but neither would be ideal I guess.

Semper-Viventem, 2024-01-22:
Hmm, I changed the screenshots for Google Play first and this aspect ratio was fine. Let me redo it on another emulator.

Semper-Viventem, 2024-01-22:
Google Play still has a recommendation to provide 16:9 screenshots but accepts 20:9 as well. My point is that most modern smartphones have taller screens here are a few examples of 2023 smartphones:

* Pixel 8: `20:9`
* Galaxy s23: `19.5:9`
* Pixel 7: `20:9`
* Asus Zenphone 10: `20:9`
* OnePlus 11 Pro: `20.1:9`
* Galaxy Fold 5: `22:9`

The screenshots I've added in the metadata are 20:9 exactly, It is pretty close to 2:1 as you mentioned above. If you need exactly 16:9, I can provide it for you outside of the `phoneScreenshots` metadata, but based on my statistics, this is a pretty rare aspect ratio in nowadays

closed 2024-01-22 by IzzySoft