BLE-Research-Group/MetaRadar #101: What for is READ_EXTERNAL_STORAGE needed?

IzzySoft, 2024-01-22:
My scanner just got some improvements, and for today's release reported `READ_EXTERNAL_STORAGE` as sensitive permission (plus a few others which were immediately clear to me and already got added to your app's allow-list). Can you please clarify what this permission is needed for? With your app targeting Android 14 and needing at least Android 10 to run, this looks a bit confusing to a non-developer like me :wink:

Thanks in advance!

Semper-Viventem, 2024-01-22:
Hey @IzzySoft. Good catch. I added this permission almost a year ago to support database backups and restores from the external storage and this permission shouldn't be a surprise for the particular release today. But it turned out that I don't need to access the whole storage (as before) because for now the app can get read/write access for particular files from the system. I'd add that `READ_EXTERNAL_STORAGE` [has no effect since Android 13](https://developer.android.com/reference/android/Manifest.permission#READ_EXTERNAL_STORAGE) and we can remove it.

Thank you for noticing, you can find the manifest update in this PR: https://github.com/Semper-Viventem/MetaRadar/pull/102

closed 2024-01-22 by Semper-Viventem

IzzySoft, 2024-01-22:
> this permission shouldn't be a surprise for the particular release today

No, definitely not. Which is why I wrote "My scanner just got some improvements". Those were fully enabled this weekend, and thus pop up for all updates since today – which is why today's release was the first one it was noticed for.

> and we can remove it

Yay!

> you can find the manifest update in this PR

(looking) Ah. Fun fact: I'd like a reference to where `READ_INTERNAL_STORAGE` is documented. Seeing it in apps for almost 10 years now, but I never found a reference. My guess so far was that it was erroneously deduced from `READ_EXTERNAL_STORAGE`, made it into some example code, and spread from there. Honestly curious if now, after all those years, I can learn its origin from you?

IzzySoft, 2024-01-23:
> That's interesting because the permission has definitely been added in Feb 2023

At that time the [additional APK checks](https://gitlab.com/IzzyOnDroid/repo/-/issues/475) were not yet in place, so today's release was the first APK of your app that was hit by them, because the others were not subject to the then not existing checks. Only updates (and new additions) are scanned by those. I get more than enough reports already that way. Once the reports slow down, I'll most likely check the older ones, too. But then again your previous releases won't be reported as those things are already on their allow-list :wink:

> this permission has been mistakenly mentioned in some old documentation or examples and spilled out across all apps

Haha! OK, so my conclusion still holds. Yes, there was one very famous piece of sample code having that mentioned. There are [several more examples](https://android.izzysoft.de/stats.php?limit=20#stat_bperms). I wonder I didn't put this one in there… Oof, I see I should check on those unknown ones again. Some of them sound funny, too: `FOREGROUND_SERVICE_LOCATION` … wait, they really did that? WTF… [Added with sdk 34](https://developer.android.com/reference/android/Manifest.permission#FOREGROUND_SERVICE_LOCATION), wow. Lots of fun ahead updating [my own reference](https://android.izzysoft.de/applists/perms) then…

> even in [the one of the AOSP apps](https://cs.android.com/android/platform/superproject/+/master:frameworks/av/media/tests/benchmark/MediaBenchmarkTest/AndroidManifest.xml;l=23?q=READ_INTERNAL_STORAGE&sq=)

ROFL :see_no_evil: