BLE-Research-Group/MetaRadar #104: App continuously requests permissions

XRatedPoetry, 2024-01-25:
**Describe the bug**
If you do not grant the app full background and location permissions it will not stop asking. 

**To Reproduce**
Steps to reproduce the behavior:
1. install app
2. Don't grant full permissions
3. It will not stop asking.

**Expected behavior**
If a user does not grant permission to something, don't ask them again.

**Smartphone (please complete the following information):**
 - Device: milanf
 - OS: Android 13
 - Browser N/A
 - Version 22.6-beta

**Additional context**
¯⁠\⁠_⁠(⁠ツ⁠)⁠_⁠/⁠¯

Semper-Viventem, 2024-01-26:
Hello @XRatedPoetry,
Do you mean that the app will continue showing the background location when you click the "Scan" button after declining the permission or does it somehow forwards you to the permission page without a "Scan" button click?

In the case of the first scenario, this is not a bug but the only one correct behavior because the main and only functionality of the app is scanning the BLE environment in the background for a long period of time. And according to the[ Android security standard](https://developer.android.com/develop/connectivity/bluetooth/bt-permissions), the Bluetooth scan requires location permission, so the background scan requires background location permission.

XRatedPoetry, 2024-01-26:
I'd really rather not revoke permissions every time I use the app. You can
at least scan if you don't grant the app background access, but would it be
possible to use the app with "location while in use only"
permissions? That's the one that immediately shunts you back to permission,
if that's intended then I don't think i'll be using this app unfortunately.

Semper-Viventem, 2024-01-26:
I'm sorry to hear that. I can consider running a scanner when the app is opened based on the "While using the app" location permission. Unfortunately, it is not possible to run in a background without background location permissions due to the android limitations.

closed 2024-01-26 by XRatedPoetry

ikanakova, 2024-02-15:
Couldn't we make it an option if the user wants to use it as a background app, and only those who want to use it in the background have to enable full location permissions?

ckuethe, 2024-03-03:
> @Semper-Viventem Couldn't you make it an option if the user wants to use it as a background app, and only those who want to use it in the background have to enable full location permissions?

Came here to make this very request. I'd like to see a way to use the app either as a one-off scan here and now, or as my pocket bluetooth wardriver.

gilcu3, 2024-04-27:
I have exactly the same request. The app would still offer valuable functions even if only used for a few minutes in the foreground.

jayb-g, 2024-07-17:
> @Semper-Viventem Couldn't you make it an option if the user wants to use it as a background app, and only those who want to use it in the background have to enable full location permissions?

I think it can be better.

> In the case of the first scenario, this is not a bug but the only one correct behavior because the main and only functionality of the app is scanning the BLE environment in the background for a long [...]

@Semper-Viventem I have seen this reply on multiple issues. Came here for same to check if other people have the same issue having their location accessed, say every few mins if scan frequency is higher. This ultimately leads to very high battery usage and not desired. According to the link you always refer to [bt-permissions](https://developer.android.com/develop/connectivity/bluetooth/bt-permissions) itself indicates ACCESS_FINE_LOCATION is required only in case of Android 11 or lower. Also see [this](https://9to5google.com/2021/05/19/android-12-nearby-permission/) 

I already use Apps like Gadgebride and Garmin Connect, Amazon Music, all of them have nearby devices permission, But they work fine without allowing location permission ever. ie they automatically detect and connect to watch/preferred speaker when nearby without having to access location.

All this to say, that it is indeed possible to be able to use Nearby permission without location. Besides, for Android 11 or lower, app has to have the location permission granted but not really need to use it _if user wants_ to use BLE Radar as just that - RADAR which shows nearby devices without need to know their GPS location. In general, Radar shows relative position and I understand that mapping devices is a stated feature but it can be made optional only for those who want to use it. Rest of us can use it as just RADAR. 

Don't get me wrong. I love the App. It's just amazing. But constant(whatever BLE scan frequency you have set) location access in the background drains my battery fast and makes the app and features useless.