BLE-Research-Group/MetaRadar #105: Map Location History not working

GR33NMAN639, 2024-02-06:
When detecting a Bluetooth signal for an extended period of time and having moved around significantly over the course of a day or two days clicking on that Bluetooth device profile and looking at the map that says location history where you can choose a day a week a month all time it doesn't matter which one I choose none of them work and none of them show the location history for the Bluetooth device on the map it just says no location history can be shown for this device under any of the choices for lengths of time that doesn't seem to be any indication of the map having any sort of history for locating that device over the course of any period of time even though it says that I've been tracking that device with me for over 23 hours

Semper-Viventem, 2024-02-23:
Hey, is this issue still relevant? I cannot reproduce such a problem. Please be sure that the location is available for the app (you can check it in the app settings location block if the scanner is turned on, there should be the latest coordinates). Also "GPS Only" mode might affect the quality of the signal.

Semper-Viventem, 2025-01-25:
I'll close the issue since I can't reproduce it. If you still observe this in the new versions please feel free to reopen.

closed 2025-01-25 by Semper-Viventem

GR33NMAN639, 2025-01-25:
What was this in regards to?