BLE-Research-Group/MetaRadar #106: BLE Service Error BLE Scan Fail

Leighlees, 2024-02-08:
**Describe the bug**
Scan feature no longer works, scan fails

**To Reproduce**
Steps to reproduce the behavior:
1. Open the app
2. Click on 'Scan'
3. No results show
4. Click on Journal tab
5. Service Errors appear multiple times

**Expected behavior**
Scan works and devices show

**Screenshots**
[image]
[image]

**Smartphone (please complete the following information):**
 - Device: Motorola Razr 40 ultra
 - OS: Android 13
 - Browser: Chrome 
 

**Additional context**
Was working perfectly to behind with. First error message ever was that Bluetooth was off.

Had profile detection once after that, then multiple error code 2. Profile detections once a day for next 3 days but nothing but error code 2 since. Scan just worked in the background, does not run anymore.

Have cleared out garbage and backed up.

Have not yet reinstalled app.

Semper-Viventem, 2024-02-08:
Hey, oh it looks like the issue I know. Could you try to reboot your phone and check if the issue is still appearing?

Leighlees, 2024-02-09:
> Hey, .oh it looks like the issue I know. Could you try to reboot your phone and check if the issue is still appearing?

Interesting, I had done this a couple of times with no success, but it has reset something this time as it appears to be scanning now. Thank you

Semper-Viventem, 2024-02-10:
Good to hear it has helped. This issue is mostly related to the Android platform logic but not the app itself. Android has a limit of 32 active registered BLE adapters across all apps and some apps might utilize the whole BLE resources (it might be even the system itself) and don't clear the adapter. The only possible solution is to relaunch the device, unfortunately.

closed 2024-02-10 by Semper-Viventem