BLE-Research-Group/MetaRadar #115: Add feauture to remove created tags, not only create them.

Username1111111111, 2024-02-21:
The ability to remove created tag seems missing. 
**NOT** to detach tag from certain device, but to completely remove tag from created tags, generaly.

Semper-Viventem, 2024-02-21:
There is no way to remove tags that have been ever used. A tag is just a string, if a tag is not applied to a device then it does not affect anything. It presents only in a tags cache on the "Select Tag" dialog. Why do you need to remove tags from the cache?

Semper-Viventem, 2024-02-21:
> Also it leaves no room for mistakes while creating tag

Tags are just strings. If you made a mistake in the tag, you can just create a new one. It would be better to remove tags from the cache anyway, let's see what can we do with this.