BLE-Research-Group/MetaRadar #116: Add ability to scan for devices without turning on location and cell network.

Username1111111111, 2024-02-21:
The location and map are not needed generally to scan nearest devices. Sometimes it hurts the privacy and completely not needed. In case of creating map locations leave turning ON as required. Just not to force user to turn it on.

Username1111111111, 2024-02-21:
I don't know much about Android privacy rules, but it seems like google forces you as developer to use the location permission otherwise bluetooth API won't work at all, right? Or is this only related to not being able to publish app on google store (not F-droid) if you violate those rules in your app?

Semper-Viventem, 2024-02-21:
@Username1111111111  This is the system-level policy. The app cannot have access to Bluetooth without the location permission. It makes sense because you can heuristically detect the approximate location based on the Bluetooth and Wi-Fi environment. Some apps did that before.

Semper-Viventem, 2024-02-21:
Also the offline mode will be available since the version `23.0-beta`: https://github.com/Semper-Viventem/MetaRadar/issues/112#issuecomment-1957586834

jayb-g, 2024-07-17:
> Talking about location permission - it is not possible to work with Bluetooth without location permission based on the [Android privacy rules](https://developer.android.com/develop/connectivity/bluetooth/bt-permissions). Also, the main purpose of the app is to analyze device location as well. So this permission will not be removed or disabled.

@Semper-Viventem I have seen this reply on multiple issues. Came here for same to check if other people have the same issue having their location accessed, say every few mins if scan frequency is higher. This ultimately leads to very high battery usage and not desired. According to the link you always refer to [bt-permissions](https://developer.android.com/develop/connectivity/bluetooth/bt-permissions) itself indicates ACCESS_FINE_LOCATION is required only in case of Android 11 or lower. Also see [this](https://9to5google.com/2021/05/19/android-12-nearby-permission/) 

I already use Apps like Gadgebride and Garmin Connect, Amazon Music, all of them have nearby devices permission, But they work fine without allowing location permission ever. ie they automatically detect and connect to watch/preferred speaker when nearby without having to access location.

All this to say, that it is indeed possible to be able to use Nearby permission without location. Besides, for Android 11 or lower, app has to have the location permission granted but not really need to use it _if user wants_ to use BLE Radar as just that - RADAR which shows nearby devices without need to know their GPS location. In general, Radar shows relative position and I understand that mapping devices is a stated feature but it can be made optional only for those who want to use it. Rest of us can use it as just RADAR.

Semper-Viventem, 2024-07-17:
Hey @jayb-g , thank you for your noticing. Your proposal sounds reasonable, I'll try to support optional location permission

Semper-Viventem, 2024-07-18:
@jayb-g the problem is that phone acceleration sensors could consume the battery even more than location. Also I'm not completely sure that it doesn't require any access as well.

Semper-Viventem, 2025-01-24:
Hey, I've made background location optional since version `0.27.0-beta`, now you can use the scanner with permission level `When I use the app`. In the general case, it even works for background activities, but sometimes the system just kills the service or restricts BLE access in the background, so an all-time location is still preferable for background observation, but not necessary.

Semper-Viventem, 2025-03-17:
@jayb-g It works in the background without all-time location permission, but there is no guarantee that the system will not kill the service or restrict BLE access. It might also depend on the device manufacturer.