BLE-Research-Group/MetaRadar #124: Add feature notification on event

kukevarius, 2024-03-04:
I would like to see a feature where you could make it so that for example when a device has been following you for 5 minutes the app can make a custom notification.

Semper-Viventem, 2024-03-04:
Hey, could you elaborate on this topic, please? For now you can create a new radar profile with the "Following" filter and get notification about this profile. The following filter supports a flexible duration threshold. Or WDYM under a custom notification?

kukevarius, 2024-03-04:
I mean that there is no option to set a customized notification for a radar profile when some conditions match.
for example when you're being followed i would like to be able to receive a customised notification from the app, so i can make the notification say like what device name is following me and for how long and that sort of stuff.

also i have tested the 'following' radar profile but i have never gotten a notification from it.

Semper-Viventem, 2024-03-04:
>  so i can make the notification say like what device name is following me and for how long and that sort of stuff

Oh, now I see. Good point

> also I have tested the 'following' radar profile but i have never gotten a notification from it.

It might depend on following duration you have set. And you should consider that some devices have a mechanism to randomize their ID once in a particular period. For example, iPhones change their BLE addresses every 10 minutes so it is difficult to track them. We consider "following" if the device moves with you for more than 300 meters for the past N minutes. Also, I'd recommend checking the power saving mode for your phone and disabling restrictions for the radar app because the system could limit Bluetooth scanning if the screen is disabled due to power saving.