BLE-Research-Group/MetaRadar #141: Map showing devices around me and where I've seen them before

dantecatalfamo, 2024-05-14:
**Is your feature request related to a problem? Please describe.**
If I have many devices around me, it is very time-consuming to tap on each device and look at its location history.

**Describe the solution you'd like**
I would like to have a view showing my device's location history, the devices around me, and where I have seen them. Instead of having each ping be a bubble like the current device map, each device could have a line connecting all the ping locations.

**Describe alternatives you've considered**
There is another app called rfparty that does something similar, though it has no way to export data lacks a lot of the features present in BLE Radar

Cessna207, 2024-06-01:
I'm trying to activate the scanner nothing to start the process I don't see nothing to start the app.

Semper-Viventem, 2025-01-25:
Good idea. History points now supports both markers and path since the app version `0.27.0-bets`

closed 2025-01-25 by Semper-Viventem

dantecatalfamo, 2025-02-14:
@Semper-Viventem Thanks a lot! The path view makes it much easier to see where things have been.

I think i may have worded my issue poorly though. What I meant was, it would be nice to have one map that shows the path of my phone, and **_all_** bluetooth devices.

The path for each bluetooth device could be a different color, and the path of the phone could be extra thick, for example. Maybe there could be a filter to only show devices that were near you for more than 10 minutes, or devices that followed you over X number of kilometers.

The problem I have is that I cannot get any real information by looking at the list of 30 bluetooth devices around me, because I have no idea where I've seen them before without clicking on each of them, and with many devices, it's not practical to actually check them all.