BLE-Research-Group/MetaRadar #142: BLE Journal

learning277, 2024-06-17:
The journal on the ble radar is not working. 
This a pixel 3xl version 14

lu-ckystars, 2024-12-08:
i have also noticed this issue on a pixel 5a

Semper-Viventem, 2025-02-03:
Hey @cactushydrocodone , does it work if you wait long enough? I suspect it might be an issue with the loading stuff, can you see the progress bar displayed?

Semper-Viventem, 2025-02-03:
Yeah, it might not be very clear. The journal is needed to keep the history of radar detections and reported system errors. I think some description is needed there

Semper-Viventem, 2025-02-03:
> I might need to add: Ive installed the apk from github, as the play store says the app is not compatible with my device, whatever that means

Could you share some details about your device like the model, android version etc? I need to check the Google Play page configuration, if the apk runs on your device, Google Play should not limit it by itself.

Semper-Viventem, 2025-02-03:
Also, I'd recommend using release versions instead of debug ones, they are more stable and consume less memory.