BLE-Research-Group/MetaRadar #175: Very sluggish

Taxit0m, 2025-03-22:
App is chugging, also it rarely identifies a device, every field for 98% of devices value is n/a, only displays address.  Only manufacturers it recognizes are apple, Samsung, and very few others.  I like this application, it used to have a ton of disability and did not bog or slow the device down.  After the latest update it continually asks to test the permission for always on location, even when already granted.  Other applications are able to identify nearly every device, a database must surely already exist it can utilize, maybe just not while running in offline mode?

Semper-Viventem, 2025-03-22:
Hey @Taxit0m , yeah I was experimenting with graphic effects last releases, it could affect performance. It was fine on my device, but could you specify your model? About manufacturers, it's a good catch, probably the database is outdated.

alberic89, 2025-04-07:
Same performance issue, the entire device is quite unusable when the app is on screen.
Device: Samsung galaxy A15

vanitasvitae, 2025-04-11:
Can confirm very poor UI performance (single digit fps) ony Fairphone 3+.

Semper-Viventem, 2025-07-30:
Hey @Taxit0m . I want to highlight that this is a free software project that I built for fun with no commercial expectations and obligations. As it's distributed under GPL3 license please feel free to fork and contribute, but at the moment I don't have time and motivation to update it