BLE-Research-Group/MetaRadar #176: Radar Profiles on Device List.

AJolly, 2025-03-24:
It would be nice if we could have the radar profiles that we create the active on the device scanning list. Also if we could easily blacklist/whitelist a tag on the device list. 

(Right now I'm creating radar profiles, or manually tagging certain Bluetooth devices, and would like to be able to hide them from the device scanning list or only specifically see those. I'm mostly using this as a way to help me find lost devices like hearing aids or a remote, but unless I'm looking for that specific device I don't need shown on the view. Right now I create a number of custom filters but those take longer to create and also just show up as custom filter at the top)

Semper-Viventem, 2025-03-24:
That's a good feature to have! I think there were mentions of this before, I'll definitely add it to the roadmap!

AJolly, 2025-03-28:
> That's a good feature to have! I think there were mentions of this before, I'll definitely add it to the roadmap!

Thanks! Or if theres a better workflow, let me know.